HackerShackOfficial/Smart-Mirror #15: Fullscreen

Warm4kerT, 2016-10-16:
I am using an raspberry screen for this project and when i run the python file just the clock and date appeared on the screen. Can you help me with this problem?

atainter, 2016-10-16:
Hmm works for me. Do you have all of the config values set in the smartmirror.py file? What is your country code? You might need a special country code for news. 
[image]

Warm4kerT, 2016-10-17:
Yes i set all config values. I test it on my laptop and worked just fine. Does exist any way that i can adjust the resolution on the .py file?

atainter, 2016-10-17:
It should autoscale to the correct resolution. Is your pi connected to the internet?

Warm4kerT, 2016-10-17:
yes it is connected. the py file does not correct the resolution.

smileyn64, 2016-10-17:
Hi,
Maybe you can try to change the background color of each frame with a different color.
it will allow you to see if all the frames are shown and maybe empty of text.

Here the modification for the backgroung, top and bottom frame:
```
class FullscreenWindow:
...
        self.topFrame = Frame(self.tk, background = '**yellow**')
        self.bottomFrame = Frame(self.tk, background = '**green**')
```

Also, are you hitting 'Enter' to switch to the fullscreen display?
I smell a minimum resolution 'non full screen' superior to the piTFT display.
When switching to full screen, it should force the scaling.

Otherwise there is a minimum resolution requirement to this project to be defined.

Warm4kerT, 2016-10-17:
yes i'mnhitting enter. thanks i will try that

atainter, 2016-10-17:
Sorry, you're right. It doesn't correct resolution, it auto scales to the screen size if you press enter. Can you post a screenshot? That might help with debugging.

Warm4kerT, 2016-10-17:
[image]

atainter, 2016-10-17:
Okay, this is definitely a scaling/layout issue. The clock component fills the whole space so the other components can't be displayed. I'll parameterize the text size tonight so that it can be changed to scale on small screens.

Warm4kerT, 2016-10-17:
Thanks man. You are the man

atainter, 2016-10-20:
Sorry, haven't been able to get to this. Will comment when updated.

atainter, 2016-10-22:
Pull the latest changes. I added parameters for font sizes.

Warm4kerT, 2016-10-22:
Thanks a lot i will test it

Warm4kerT, 2016-10-23:
Well i test it and it wasn't enough to show everything in the screen. Can i reduce the space between the clock and the weather. Is there any way that i can remove the News, because in my country isn't supported.
Thanks a lot for the help.

smileyn64, 2016-10-23:
For moving the frames you can play with the 'padx' & 'pady' parameters in the class FullScreenWindow.
If you want to remove the news, try by commenting (with # sign) the lines for news frame in the same class.
It should work.

atainter, 2016-10-23:
Yes, there are quite a few padding options that need to be adjusted. I'm not sure if I can parameterize all of those.

closed 2016-11-15 by atainter

bergsm, 2017-12-12:
Adjust the padding options doesn't seem to work. Once the frame moves too far one way (like the towards the center) it gets cut off.