HackerShackOfficial/Smart-Mirror #16: cannot get weather

Ctian011597, 2016-10-18:
please help!
its working on previous days
now i cant run "smartmirror.py"
there's something showing like;

File: "smartmirror.py', line 190
print "Error: %s. Cannot get weather." % e

i'm still newbie sorry.

smileyn64, 2016-10-18:
Hi,
did you get the last update of the files ?
previsouly the forcast was pulled from the website forecast.io that is now know as darksky.net.

Best is to proceed to re-installation of the smart mirror program to get the last features commited.

You can also modify the following line:
`weather_req_url = "https://api.forecast.io/forecast/%s/%s,%s" % (weather_api_token, lat,lon)`
as follows:
`weather_req_url = "https://api.darksky.net/forecast/%s/%s,%s" % (weather_api_token, lat,lon)`

This modification is already implemented in the last revision of the program, so I insist on proceed to update.

Ctian011597, 2016-10-18:
thank you @smileyn64  :)

but do you think it is possible to set my own location? for weather darksky forecast?

smileyn64, 2016-10-18:
Yes it's possible,

You can change the values of the variables **_lat_**, **_lon_** and **_location2_**.

**_lat_** shall be equal to your lattitude
**_lon_** shall be equal to your longitude
**_location2_** is text that will be displayed as Weather location

example:
```
lat = -6.5191
lon = 105.8078
location2 = "Jakarta, BT"
```

Be carefull and keep track of your modifications in case you want back the location IP based.

Ctian011597, 2016-10-18:
i'll test it later :)
can i ask you again later if it doesn't work? please?

smileyn64, 2016-10-18:
just write down here, I will reply if I can help ;)

atainter, 2016-10-20:
@smileyn64 thanks for helping. Closing this issue.

closed 2016-10-20 by atainter