HackerShackOfficial/Smart-Mirror #19: Something is wrong Please HELP!

HumzahF, 2016-10-20:
When I try to run the code, I get this error and it does not display the weather but displays everything else:

  File "smartmirror.py", line 147, in get_weather
    weather_obj = json.loads(r.text)
  File "/usr/lib/python2.7/json/__init__.py", line 338, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 366, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 384, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
Error: No JSON object could be decoded. Cannot get weather.

atainter, 2016-10-20:
Make sure you add your weather token

HumzahF, 2016-10-20:
Yes i made sure to do that many times

atainter, 2016-10-20:
add a line print `r.text` before `weather_obj = json.loads(r.text)`

HumzahF, 2016-10-20:
I put (print r.text) in the place you mentiobed but its still not working

atainter, 2016-10-20:
What does it print?

HumzahF, 2016-10-20:
It prints nothing

atainter, 2016-10-20:
Can you print the request url for weather, then paste it in your browser?

atainter, 2016-10-20:
also, did you use `(print r.text) ` or `print r.text`

atainter, 2016-10-20:
Did you get your api token through darksky.net?

HumzahF, 2016-10-20:
Yes i did. Also i get a 400 bad request when i copy paste the url into my
browser

atainter, 2016-10-20:
Can you remove the token and post the url here?

HumzahF, 2016-10-20:
sorry for the lateness but here is the complete code

 # get weather
            weather_req_url = "
https://api.darksky.net/forecast/%s/%s,%s?lang=%s&units=%s" %
(weather_api_token, lat,lon,weather_lang,weather_unit)
            r = requests.get(weather_req_url)
            print r.text
            weather_obj = json.loads(r.text)

thank you so much

King3060, 2016-10-20:
I am also getting this error. I have added print r.text but it says Error: No JSON object could be decoded. Cannot get weather. also i seem to be getting American news. Is the news code uk, gb or en? Thanks a lot for your help.

atainter, 2016-10-22:
I added parameters for latitude and longitude. Try setting both of those and trying again.

King3060, 2016-10-22:
where did you add them?

atainter, 2016-10-22:
Top of the file. By the params for weather token.

King3060, 2016-10-22:
also would it help if i opened the file with python 3?

King3060, 2016-10-22:
it seems that the code isnt even taking weather_api_token variable into account. I just left it blank and it came up with the usual error

atainter, 2016-10-22:
I use python 2.7 exclusively, but it might work with 3

atainter, 2016-10-22:
Strange. It works for me. Maybe you didn't copy the token correctly? Also, are you wrapping your token in quotes?

King3060, 2016-10-22:
For python 3 it comes out with a syntax error

King3060, 2016-10-22:
weather_api_token = 'ab3ecce009842378d898bcfc0d1b7270'

atainter, 2016-10-22:
wfm 
[image]
[image]

King3060, 2016-10-22:
https://api.darksky.net/forecast/%s/%s,%s?lang=%s&units=%s this link has a 400 error

King3060, 2016-10-22:
lol its not 68 degrees here

King3060, 2016-10-22:
im on raspbian would that make a difference?

atainter, 2016-10-22:
That's my location.

You need to replace the `%s` values in the string. https://api.darksky.net/forecast/ab3ecce009842378d898bcfc0d1b7270/37.774929,-122.419416?lang=en&units=us

Where 37.774929,-122.419416 are the latitude and longitude respectively.

King3060, 2016-10-22:
is that the link from under the api? because ive tried putting that in already

King3060, 2016-10-22:
Error: not all arguments converted during string formatting. Cannot get weather.

atainter, 2016-10-22:
When did you get this error?

atainter, 2016-10-22:
I constructed the URL

King3060, 2016-10-22:
after i copied the link up there. and put it in here: weather_req_url = "https://api.darksky.net/forecast/ab3ecce009842378d898bcfc0d1b7270/37.774929,-122.419416?lang=en&units=us" % (weather_api_token, lat,lon,weather_lang,weather_unit)

atainter, 2016-10-22:
remove `% (weather_api_token, lat,lon,weather_lang,weather_unit)`

King3060, 2016-10-22:
it says i have a total number of calls of 16, so something must be happening

atainter, 2016-10-22:
oh, you must've reached your call limit for the hour. Try again in an hour

King3060, 2016-10-22:
you sir, are a wizard

King3060, 2016-10-22:
it works :)

King3060, 2016-10-22:
but now there is a comma infront of eng

atainter, 2016-10-22:
screenshot?

King3060, 2016-10-22:
for some reason i cant attach files here...

atainter, 2016-10-22:
drag and drop them into the textbox

King3060, 2016-10-22:
also i cant take a screen shot while the window is open...

atainter, 2016-10-22:
Not sure why the screenshot wont work.

It probably can't find your location from ip for some reason. That is separate from the weather api, but sends the lat/long coordinates to the weather api. You could hard code the location name if you want. Just turn set `location2`

atainter, 2016-10-22:
For example `location2 = San Francisco, CA`

atainter, 2016-10-22:
Also, you have to make sure you update the url with your correct lat/long

King3060, 2016-10-22:
thanks a lot. it works perfectly now :) Can i ask you a favour?

King3060, 2016-10-22:
just a little one?

closed 2016-10-23 by atainter

Projectchotu, 2017-04-13:
i am getting this error 

raceback (most recent call last):
  File "/home/pi/123.py", line 235, in get_headlines
    headline = NewsHeadline(self.headlinesContainer, post.title)
  File "/home/pi/123.py", line 248, in __init__
    image = Image.open("assets/Newspaper.png")
  File "/usr/lib/python2.7/dist-packages/PIL/Image.py", line 2251, in open
    fp = builtins.open(fp, "rb")
IOError: [Errno 2] No such file or directory: 'assets/Newspaper.png'
Error: [Errno 2] No such file or directory: 'assets/Newspaper.png'. Cannot get news.

please help

ashish7788, 2017-04-14:
Check in your directory assets folder in which  newspaper.png image is
prsent or not..

Projectchotu, 2017-04-18:
there is no such folder assets in my directory.....

Projectchotu, 2017-04-18:
please help......

Projectchotu, 2017-04-18:
i created a folder assets with Newspaper.png but still the same error.....
/home/pi/MagicMirror/assets

kushishristi, 2018-05-24:
hi guys i am also getting same error in smart mirror master(No json object could be decoded.cannot get weather)... 
[image]
...please help me it's urgent

kushishristi, 2018-05-24:
[image] this my code. please help me.

kushishristi, 2018-05-25:
[New Microsoft Office Word Document.docx](https://github.com/HackerShackOfficial/Smart-Mirror/files/2038208/New.Microsoft.Office.Word.Document.docx)
please help me any one i am getting error.

LLeo007, 2019-04-27:
Can I get fresh codes of python 3 for smart mirror?

MudassirKazmi, 2019-04-27:
Enter a fresh ip token and try again..