HackerShackOfficial/Smart-Mirror #23: Error when running smartmirror.py

kenthmf, 2016-10-23:
Cloned the files.
Updated the IP and Weather API token then saved the smartmirror.py.
Confirmed following the procedure as instructed in the video and tried download the file again but having same issue.

Having following error when running:

pi@kenthpi:~/Smart-Mirror $ python smartmirror.py
Traceback (most recent call last):
  File "smartmirror.py", line 330, in <module>
    w = FullscreenWindow()
  File "smartmirror.py", line 297, in __init__
    self.tk = Tk()
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1813, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: no display name and no $DISPLAY environment variable
pi@kenthpi:~/Smart-Mirror $

King3060, 2016-10-23:
R u connected to the internet?

kenthmf, 2016-10-23:
Thanks.
I added the Latitude, Longitude value to the script and rerun the script then it works.
Seems like the  Latitude, Longitude value cannot be set to None.

In my case, no need to tweek the URL.

King3060, 2016-10-23:
So it works 100% now?

Ctian011597, 2016-10-24:
how to set the Latitude, Longitude value?
please help me @kenthmf

kenthmf, 2016-10-24:
you can find following 2 lines in the script.
---------------------------------------------------------
latitude = None # Set this if IP location lookup does not work for you (must be a string)
longitude = None # Set this if IP location lookup does not work for you (must be a string)
---------------------------------------------------------

go to any website which offer the latitude, longtitude value such as 
http://www.latlong.net/
http://www.findlatitudeandlongitude.com/

get the value and replace the 'None' with the value.

closed 2016-10-24 by kenthmf