HackerShackOfficial/Smart-Mirror #24: Not really an issue. But don't know where to ask.

King3060, 2016-10-23:
I was just wondering if any one could help me. I added random messages in the middle of the screen using the calendar class as a template. But I can't seem to figure out how to update it so it prints a new message after a couple of minutes.
[image]
.

Ctian011597, 2016-10-30:
mr. @King3060 
please send me your codes.
thank you let us make this happen 👍 

cristianbalunan@gmail.com

closed 2016-11-15 by atainter

liquid1, 2016-12-25:
mr. @King3060  please send me your code because i am facing issues in weather, i sort out the errors in news but unable to debug in the weather section.
thanks
liquid7127@gmail.com

liquid1, 2016-12-25:
i figured out the errors.
no need to send your code.
thanks