HackerShackOfficial/Smart-Mirror #28: Can't get weather data.

cedson61, 2016-11-01:
I'm not sure if this is my issue, or your's, but I can't get any weather data.

Here is my issue:

Traceback (most recent call last):
  File "smartmirror.py", line 130, in get_ip
    ip_json = json.loads(req.text)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/__init__.py", line 338, in loads
    return _default_decoder.decode(s)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/decoder.py", line 366, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/decoder.py", line 384, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
Traceback (most recent call last):
  File "smartmirror.py", line 143, in get_weather
    location_obj = json.loads(r.text)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/__init__.py", line 338, in loads
    return _default_decoder.decode(s)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/decoder.py", line 369, in decode
    raise ValueError(errmsg("Extra data", s, end, len(s)))
ValueError: Extra data: line 1 column 5 - line 2 column 1 (char 4 - 19)
Error: Extra data: line 1 column 5 - line 2 column 1 (char 4 - 19). Cannot get weather.

It has to do with not getting the data from darksky, I'm pretty sure.

atainter, 2016-11-15:
you've added your api token, correct?

delahoya1, 2016-11-24:
i have the same issue i cant get weather and this is the error code
Traceback (most recent call last):
  File "smartmirror.py", line 158, in get_weather
    weather_obj = json.loads(r.text)
  File "/usr/lib/python2.7/json/__init__.py", line 338, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 366, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 384, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
Error: No JSON object could be decoded. Cannot get weather.

odblue, 2016-11-24:
i have the same problem the secret code is the api token correct but it wont read into my mirror

King3060, 2016-11-28:
I had this issue too look in the closed issues section. might help.

delahoya1, 2016-11-28:
what is the closed issues section?

King3060, 2016-11-30:
go on issues. then press closed.

J1306, 2017-02-05:
make sure you've deleted the < > Parenthesis when inputing the api token

dmrowe84, 2017-03-10:
i get the same error as @delahoya1 
heres my .py file
[image]

tomellis91, 2017-04-05:
Dmrowe84 you have the whole URL in your token. '5fexxxxxxx' is the only part you require.

ashish7788, 2017-04-05:
This Is my Token URL

"
http://api.openweathermap.org/data/2.5/weather?q=Ahmedngar,IN&appid=b26b5de6257ed4e1a3fb25ab840e3684&units=metric
"

This is My Token : b26b5de6257ed4e1a3fb25ab840e3684
I have try this weather api code in separate it works but in smart mirror
this shows error , I will send that snapshot also..

tomellis91, 2017-04-05:
http://api.openweathermap.org/data/2.5/weather?q=Ahmedngar,IN&appid=b26b5de6257ed4e1a3fb25ab840e3684&units=metric is the full call. The code handles that all you need to insert is: b26b5de6257ed4e1a3fb25ab840e3684

pacha5065, 2017-05-17:
make sure that you provide the correct value for following variables, when you encounter this issue: 

weather_api_token ==> Provide only the secret token that you get from forecast website but not the full URL.
weather_language ==> you can get these language codes from the darksky forecast website itself, it is "en" in my case.
weather_unit ==> even this will provided in the darksky forecast website, i choose "auto" in this case.
longitude and latitude were part of the TOKER URL get it from there and provide it here.

It should work fine now.

ShraddhaDevaiya, 2017-06-27:
I have added whether api token but then also it shows this type of error.how can I solve?

_Sent from my Xiaomi Redmi 3S using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_