HackerShackOfficial/Smart-Mirror #30: Missing Assets Error

n4c0n, 2016-11-04:
Sometimes the program works nearly perfectly, as can be seen in my previous post. However, sometimes, I get the following output in terminal an on the screen when run "python Code/Smart-Mirror/smartmirror.py ":

```
Traceback (most recent call last):
  File "Code/Smart-Mirror/smartmirror.py", line 174, in get_weather
    image = Image.open(icon2)
  File "/usr/lib/python2.7/dist-packages/PIL/Image.py", line 2251, in open
    fp = builtins.open(fp, "rb")
IOError: [Errno 2] No such file or directory: 'assets/PartlyMoon.png'
Error: [Errno 2] No such file or directory: 'assets/PartlyMoon.png'. Cannot get weather.
Traceback (most recent call last):
  File "Code/Smart-Mirror/smartmirror.py", line 238, in get_headlines
    headline = NewsHeadline(self.headlinesContainer, post.title)
  File "Code/Smart-Mirror/smartmirror.py", line 251, in __init__
    image = Image.open("assets/Newspaper.png")
  File "/usr/lib/python2.7/dist-packages/PIL/Image.py", line 2251, in open
    fp = builtins.open(fp, "rb")
IOError: [Errno 2] No such file or directory: 'assets/Newspaper.png'
Error: [Errno 2] No such file or directory: 'assets/Newspaper.png'. Cannot get news.
```

[image]

Can anyone help?

King3060, 2016-11-04:
I had this issue too. Make sure there is only one file called smartmirror.py because when you edit with nano you might save it in the wrong place.

n4c0n, 2016-11-05:
I solved my problem. It didn't have anything to do with multiple files of the same name. It was about running the script while in another directory in the terminal. Seems I should have made my way to Code/Smart-Mirror before executing the command "python smartmirror.py" I'm guessing the script was looking for the assets in the folder in which I made the call. So, it was looking for an assets folder in the home directory, when it should have been looking for them in the assets sub-directory of Smart-Mirror.

Alternatively, and what I ended up doing was changing the code:

`assets/WHATEVERWEATHER.png`

to

`SOMEHOMESUBDIRECTORY/SUBDIR/SUBDIR/...SUBDIR/assets/WHATEVERWEATHER.png`

I hope that makes sense.

closed 2016-11-05 by n4c0n