HackerShackOfficial/Smart-Mirror #31: weather

Rimap47, 2016-11-09:
Can you please let me know why I get these error ? And on the display only date/time and news comes up, nothing more.
Traceback (most recent call last):
  File "smartmirror.py", line 158, in get_weather
    weather_obj = json.loads(r.text)
  File "/usr/lib/python2.7/json/__init__.py", line 338, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 366, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 384, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
Error: No JSON object could be decoded. Cannot get weather.﻿
Show less

atainter, 2016-11-15:
You've added your API token, correct?

Rimap47, 2016-11-16:
thanks for your replay, I figured it out yesterday. So now its works just fine. But where does it get the news from ? I would like to see local news you know.

VBQL, 2016-11-18:
You get news from a google rss feed using feedparser. To get your local news, find any rss feed and replace the google one. Also, you might want to remove the variables for country and only have feedparser read one rss.

closed 2016-11-18 by atainter

Rimap47, 2016-11-18:
Thanks for your replay VBQL, I have replaced the rss feeds but gets following error:

Traceback (most recent call last):
  File "smartmirror.py", line 231, in get_headlines
    headlines_url = "www.nrk.no/nyheter/toppsaker.rss" % news_country_code
TypeError: not all arguments converted during string formatting
Error: not all arguments converted during string formatting. Cannot get news.

Which variables should I remove ?

VBQL, 2016-11-19:
You should remove the news variables that contains the country codes, that seems to mess it up for me.

VBQL, 2016-11-19:
This should be the code in the news class:
class News(Frame):
    def __init__(self, parent, *args, **kwargs):
        Frame.__init__(self, parent, *args, **kwargs)
        self.config(bg='black')
        self.title = 'News'
        self.newsLbl = Label(self, text=self.title, font=('Helvetica', medium_text_size), fg="white", bg="black")
        self.newsLbl.pack(side=TOP, anchor=W)
        self.headlinesContainer = Frame(self, bg="black")
        self.headlinesContainer.pack(side=TOP)
        self.get_headlines()
    def get_headlines(self):
        try:
            for widget in self.headlinesContainer.winfo_children():
                widget.destroy()
            feed = feedparser.parse("www.nrk.no/nyheter/toppsaker.rss")
            for post in feed.entries[0:5]:
                headline = NewsHeadline(self.headlinesContainer, post.title)
                headline.pack(side=TOP, anchor=W)

Rimap47, 2016-11-19:
Thank you VBQL, looks like that did the jobb. Can I ask you for 1 more thing ?

Regarding  the location.

When i set the latitude and longitude, the code doesnt display the location and shows the weather degree unormal-ly high. And doesnt give any error either. 

If i remove the latitude and longitude strings, then the code shows " can not pin point location" and weather degree become normal.

latitude = '60.408639' # Set this if IP location lookup does not work for you (must be a string)
longitude = '5.227588' # Set this if IP location lookup does not work for you (must be a string)

I also tried it without apostrophe (')

VBQL, 2016-11-20:
First, you might want geolocation to be obtained from google maps, also, the default location is displayed in Fahrenheit. To change in celsuis, change the code from:
return 1.8 * (kelvin_temp - 273) + 32
to
return kelvin_temp - 273

delahoya1, 2016-12-01:
riman47 how did you figure out the error?

pallense, 2016-12-01:
I downloaded the software ( package) and install them from the folder

liquid1, 2016-12-25:
rimap47?? how did you figure out the error?

Rimap47, 2016-12-26:
Which issue ?

rosegoldthakur, 2017-01-10:
@Rimap47 how did you fix the Weather issue ? explain please.

Rimap47, 2017-03-24:
"add api token" did fix the issue.

ShraddhaDevaiya, 2017-06-22:
@Rimap47 I added api token still it showing this type of error

_Sent from my Xiaomi Redmi 3S using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_