HackerShackOfficial/Smart-Mirror #32: Starting issues

peterpuku, 2016-11-15:
Can you please help me, because I get multiple errors when i try to start the program.

 File "smartmirror.py", line 330, in <module>
    w = FullscreenWindow()
  File "smartmirror.py", line 307, in __init__
    self.clock = Clock(self.topFrame)
  File "smartmirror.py", line 77, in __init__
    self.tick()
  File "smartmirror.py", line 80, in tick
    with setlocale(ui_locale):
  File "/usr/lib/python2.7/contextlib.py", line 17, in __enter__
    return self.gen.next()
  File "smartmirror.py", line 39, in setlocale
    yield locale.setlocale(locale.LC_ALL, name)
  File "/usr/lib/python2.7/locale.py", line 579, in setlocale
    return _setlocale(category, locale)
locale.Error: unsupported locale setting

warflash, 2016-11-17:
Yeah, I get the exact same error

atainter, 2016-11-18:
Did you set anything for the line `ui_locale = '' # e.g. 'fr_FR' fro French, '' as default`?

artimousgordon, 2016-12-22:
@atainter yes to 'us_US' and same errors came up

ToxicYDG, 2016-12-30:
I am having the same issue. Nothing is working.

ToxicYDG, 2016-12-30:
I figured it out, delete the '<>' from where you put in your api code

chenoi, 2017-01-28:
Please check your system locale in your machine
locale -a
Make sure your enter the correct system for example ui_locale='en_US.utf8' and this is listed in your system locales machine.