HackerShackOfficial/Smart-Mirror #33: No "Wind.png" in the assets

Zblay, 2016-11-21:
Hello, 

And thanks for this project, i love it. 

I ran into an issue today : There's no "Wind.png" in the assets so i added one to my own Smart-Mirror but thing is the Image i used is CC-BY, so the author name is to be mentionned, can anyone find a CC "Wind.png" ( or even create one ? )

Cheers

King3060, 2016-11-28:
You might be running smartmirror.py from the wrong directory. Also make sure you have not made saved other copies outside the Smart-Mirror directory.

Zblay, 2016-11-28:
https://github.com/HackerHouseYT/Smart-Mirror/tree/master/assets
I might be wrong but i see no "Wind.png" in there

atainter, 2016-11-28:
Sorry about that. Just added an icon for Wind.png.

closed 2016-11-28 by atainter