HackerShackOfficial/Smart-Mirror #34: permission denied

pallense, 2016-11-24:
Permission denied (publickey).
fatal: Could not read from remote repository.

King3060, 2016-11-28:
Try adding an SSH key. Look in Settings and SSH and GPG keys

milluhgram, 2017-01-26:
I'm trying to do the clone but I'm getting the same thing. Can someone help me?

yashvardhan10, 2017-02-06:
please help with the issue !

robozera, 2017-02-15:
same issue here... Am I so newbie for this??

argenis69, 2017-02-16:
Hi pallense did you ever find out the problem and how to fix it i have the same issue and i have no idea on what to do
ps:IM new to raspberry

markferreira01, 2017-08-01:
You need to add an ssh key to your settings in GitHub. I don't have instructions but Google has a ton of amswers

Wernervdmerwe, 2019-07-15:
The below worked fine for me: `git clone https://github.com/HackerShackOfficial/Smart-Mirror.git`