HackerShackOfficial/Smart-Mirror #38: Pixel Width

ertiming, 2016-12-17:
I'm curious how to fix this. I've tried adjusting the pad x and y but the text is somehow getting cut
[image]
[image]

markferreira01, 2017-08-01:
If you haven't figured this out yet, try adding expand =1, fill=BOTH into some of the .pack() commands for weather and news. It worked for me.

bervin61, 2017-12-27:
If anyone is still running into this issue, I came up with a fix. You'll need to adjust the hard-coded values to fit your screen

For the forecast:
```
            forecast2 = weather_obj["hourly"]["summary"]
            start=0
            while len(forecast2[start:len(forecast2)])>40 and forecast2.count(' ',start+35):
                start=forecast2.find(' ',start+35)
                forecast2=forecast2[0:start]+'\n'+forecast2[start+1:len(forecast2)]
```

And for the News stories:
```
            for post in feed.entries[1:6]:
                title=post.title
                start=0
                while len(title[start:len(title)])>100 and title.count(' ',start+90):
                    start=title.find(' ',start+90)
                    title=title[0:start]+'\n'+title[start+1:len(title)]
                
                headline = NewsHeadline(self.headlinesContainer, title)
                headline.pack(side=TOP, anchor=W)
```