HackerShackOfficial/Smart-Mirror #39: feedparser error

reg44reg, 2016-12-19:
Trying to run smart mirror and receive the following ImportError:

Traceback (most recent call last):
  File "smartmirror.py", line 12, in <module>
     import feedparser
ImportError: No module named feedparser

Any help would be great.

atainter, 2016-12-20:
`pip install feedparser` You also might want to install from requirements: `pip install -r requirements.txt`

reg44reg, 2016-12-24:
Didn't work, I'm getting the same error when I try to run the application. 
Thanks for your help with this...

bminer1, 2016-12-30:
I am having the same problem as well.
When i try to install pip install feedparser and the pip install -r requirements.txt, I get the below:

error: [Errno 13] Permission denied: '/usr/local/lib/python2.7/dist-packages/feedparser.py'

----------------------------------------
Cleaning up...
Command /usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-build-f9Y4Mv/feedparser/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-BnlX1u-record/install-record.txt --single-version-externally-managed --compile failed with error code 1 in /tmp/pip-build-f9Y4Mv/feedparser
Storing debug log for failure in /home/pi/.pip/pip.log
pi@raspberrypi:~ $ pip install -r requirements.txt
Could not open requirements file: [Errno 2] No such file or directory: 'requirements.txt'
Storing debug log for failure in /home/pi/.pip/pip.log

markferreira01, 2017-08-01:
Bminor try using Audi before the install command

harrysprogramming, 2019-02-10:
the module feedparser only supports up to python 3.2 look on their github it says that there

Aashishkumar123, 2020-09-11:
### **### feedparse is not installing , Showing this error** 

Collecting feedparser
  Downloading feedparser-5.2.1.tar.bz2 (192 kB)
ERROR: Exception:
Traceback (most recent call last):
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\site-packages\pip\_vendor\urllib3\response.py", line 425, in _error_catcher
    yield
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\site-packages\pip\_vendor\urllib3\response.py", line 507, in read
    data = self._fp.read(amt) if not fp_closed else b""
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\site-packages\pip\_vendor\cachecontrol\filewrapper.py", line 62, in read
    data = self.__fp.read(amt)
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\http\client.py", line 454, in read
    n = self.readinto(b)
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\http\client.py", line 498, in readinto
    n = self.fp.readinto(b)
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\socket.py", line 669, in readinto
    return self._sock.recv_into(b)
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\ssl.py", line 1241, in recv_into
    return self.read(nbytes, buffer)
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\ssl.py", line 1099, in read
    return self._sslobj.read(len, buffer)
socket.timeout: The read operation timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\site-packages\pip\_internal\cli\base_command.py", line 188, in _main
    status = self.run(options, args)
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\site-packages\pip\_internal\cli\req_command.py", line 185, in wrapper
    return func(self, options, args)
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\site-packages\pip\_internal\commands\install.py", line 332, in run
    requirement_set = resolver.resolve(
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\site-packages\pip\_internal\resolution\legacy\resolver.py", line 179, in resolve
    discovered_reqs.extend(self._resolve_one(requirement_set, req))
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\site-packages\pip\_internal\resolution\legacy\resolver.py", line 362, in _resolve_one
    abstract_dist = self._get_abstract_dist_for(req_to_install)
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\site-packages\pip\_internal\resolution\legacy\resolver.py", line 314, in _get_abstract_dist_for
    abstract_dist = self.preparer.prepare_linked_requirement(req)
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\site-packages\pip\_internal\operations\prepare.py", line 467, in prepare_linked_requirement
    local_file = unpack_url(
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\site-packages\pip\_internal\operations\prepare.py", line 255, in unpack_url
    file = get_http_url(
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\site-packages\pip\_internal\operations\prepare.py", line 129, in get_http_url
    from_path, content_type = _download_http_url(
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\site-packages\pip\_internal\operations\prepare.py", line 281, in _download_http_url
    for chunk in download.chunks:
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\site-packages\pip\_internal\cli\progress_bars.py", line 166, in iter
    for x in it:
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\site-packages\pip\_internal\network\utils.py", line 15, in response_chunks
    for chunk in response.raw.stream(
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\site-packages\pip\_vendor\urllib3\response.py", line 564, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\site-packages\pip\_vendor\urllib3\response.py", line 529, in read
    raise IncompleteRead(self._fp_bytes_read, self.length_remaining)
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\contextlib.py", line 131, in __exit__
    self.gen.throw(type, value, traceback)
  File "c:\users\user\appdata\local\programs\python\python38-32\lib\site-packages\pip\_vendor\urllib3\response.py", line 430, in _error_catcher
    raise ReadTimeoutError(self._pool, None, "Read timed out.")
pip._vendor.urllib3.exceptions.ReadTimeoutError: HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Read timed out.