HackerShackOfficial/Smart-Mirror #40: cant get weather?

liquid1, 2016-12-21:
Traceback (most recent call last):
  File "smartmirror.py", line 158, in get_weather
    weather_obj = json.loads(r.text)
  File "/usr/lib/python2.7/json/__init__.py", line 338, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 366, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 384, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
Error: No JSON object could be decoded. Cannot get weather.

(and i have placed the API token)

liquid1, 2016-12-25:
i figured out every error related to news and weather.
i am disappointed for your response.
 i even mailed you on gmail but even no reply from there.
Anyways thanks for this great opportunity for keeping this open source.
[image]

closed 2016-12-25 by liquid1

rosegoldthakur, 2017-01-10:
@liquid1  hey could you please explain that how you fixed that above issue ? i am struggling with the same error right now. Also is the Secret Key the API TOKEN ? Please help!!

liquid1, 2017-01-11:
send the screenshot of your weather code..

rosegoldthakur, 2017-01-11:
@liquid1 hey I have got it to work now. I was not deleting the parenthesis before and after my TOKEN. Everything is working good for me now. Thank you :)

liquid1, 2017-01-11:
no problem. if you will face any other problem feel free to ask.

seonghun0312, 2017-01-24:
hey @liquid1, I have same error. I know it's kinda late to ask, but can you fix mine too? thanx

jamesbetterman, 2017-01-25:
hey @liquid1 i know i am kinda late but do you think you could show me whats wrong? ive got screenshots. 
[image]
[image]

enkuu, 2017-02-15:
I kinda have a same problem here guys. Anyone got a solution for this?

aRandomGadjee, 2018-03-01:
@jamesbetterman Your token, You only need the token code, not the URL I believe

nfao2, 2019-11-24:
hi can you help me with my issue? i keep getting this 
in get_weather temperature2 = "%s%s" %(str(int(weather_obj['currently']['temperature'])), degree_sign) KeyError: 'currently' Error: 'currently'. Cannot get weather.