HackerShackOfficial/Smart-Mirror #41: ImageTK error

njzroadrunner, 2016-12-30:
I'm receiving the following error when trying to run python smartmiorror.py within the smart mirror folder.  

Traceback (most recent call last):
  File "smartmirror.py", line 14, in <module>
    from PIL import Image, ImageTk
ImportError: cannot import name ImageTk

Any help would be greatly appreciated. Thanks!

sid2934, 2016-12-30:
If you have not ran the pip install Pillow this will ensure you have the
library.

njzroadrunner, 2016-12-30:
Thanks for responding.  I confirmed that this was ran and installed correctly.  I am new to programming so there could be something I'm missing but I followed instructions for installing pip

mdossantos, 2017-02-18:
@njzroadrunner , at this additional line
sys.path.append("/usr/local/lib/python2.7/site-packages")
from PIL import Image, ImageTk