HackerShackOfficial/Smart-Mirror #42: Error: 'currently'

LuckyLuke009, 2016-12-31:
Hi there,

I hope anyone can help me with this. Everything works, except the wheater.
I always get this message when I start it up:
`Traceback (most recent call last):
  File "smartmirror.py", line 161, in get_weather
    temperature2 = "%s%s" %(str(int(weather_obj['currently']['temperature'])), degree_sign)
KeyError: 'currently'
Error: 'currently'. Cannot get weather.
`

I also put in the secret code as it should be.

Thanks and greets from Germany

Lukas

closed 2017-01-11 by LuckyLuke009

liquid1, 2017-01-11:
have you figure out the error or you need help?

voodoolights, 2017-01-23:
I have the same problem. could you share how you solved it please?
Thank you in advance!

nfao2, 2019-11-24:
I have the same error too!! How did you fix it?