HackerShackOfficial/Smart-Mirror #43: So close....

gbcompton, 2017-01-01:
I've followed the instruction to the letter and everything installed correctly. When trying to run smartmirror.py, I get the following:
### pi@raspberrypi:~/Smart-Mirror $ python smartmirror.py 
Traceback (most recent call last):
  File "smartmirror.py", line 330, in <module>
    w = FullscreenWindow()
  File "smartmirror.py", line 307, in __init__
    self.clock = Clock(self.topFrame)
  File "smartmirror.py", line 77, in __init__
    self.tick()
  File "smartmirror.py", line 80, in tick
    with setlocale(ui_locale):
  File "/usr/lib/python2.7/contextlib.py", line 17, in __enter__
    return self.gen.next()
  File "smartmirror.py", line 39, in setlocale
    yield locale.setlocale(locale.LC_ALL, name)
  File "/usr/lib/python2.7/locale.py", line 579, in setlocale
    return _setlocale(category, locale)
locale.Error: unsupported locale setting

Anyone know what's going on here? Thanks.

gbcompton, 2017-01-01:
I figured it out. Here is the important part of my code for anyone else having this issue:
ui_locale = 'en_US.UTF-8'
weather_api_token = 'stringofnumbersfromthedarkskysitetoken' (don't use the entire URL like I've seen some people try)
I added my lat and long in a string format where space is provided.

closed 2017-01-01 by gbcompton

corruptedash, 2018-06-03:
Thanks. It resolves the problem.