HackerShackOfficial/Smart-Mirror #44: Auto start in full screen

blackc2004, 2017-01-04:
Is there a way you can autostart the app in full screen on boot??

Zblay, 2017-01-05:
I'm interested as well, i tried sereval things, but can't get them to work properly

ric96, 2017-01-10:
Remove 
This
self.tk.bind("<Return>", self.toggle_fullscreen)
self.tk.bind("<Escape>", self.end_fullscreen)

With
This
self.toggle_fullscreen()

LReene, 2017-04-27:
Am new to this board but have been playing with Smart-Mirror for a while now.  Desperately want to get it up and installed in my bathroom mirror but STILL can't figure out how to get it to autostart the app in full screen.  I'm not sure whether I have an old version, or a new version, or what, but the text listed in this example is no where to be found in my smartmirror.py  Can someone please help?

LReene, 2017-04-27:
Ok, so I did find the above "self.tk.bind" statements (grep works wonders).  Now if I can just figure out how to make it auto launch on boot and shut it down, I will be all set.  Any helpers?