HackerShackOfficial/Smart-Mirror #45: locale.Error: unsupported locale setting

Tchev146, 2017-01-05:
Hi, i have an issue when i try to execute "sudo python ./smartmirror.py", can anyone help ? 
This is the error that's return : 

pi@SmartMirror:~/Downloads/Smart-Mirror-master$ sudo python smartmirror.py 
Traceback (most recent call last):
  File "smartmirror.py", line 330, in <module>
    w = FullscreenWindow()
  File "smartmirror.py", line 307, in __init__
    self.clock = Clock(self.topFrame)
  File "smartmirror.py", line 77, in __init__
    self.tick()
  File "smartmirror.py", line 80, in tick
    with setlocale(ui_locale):
  File "/usr/lib/python2.7/contextlib.py", line 17, in __enter__
    return self.gen.next()
  File "smartmirror.py", line 39, in setlocale
    yield locale.setlocale(locale.LC_ALL, name)
  File "/usr/lib/python2.7/locale.py", line 579, in setlocale
    return _setlocale(category, locale)
locale.Error: unsupported locale setting
pi@SmartMirror:~/Downloads/Smart-Mirror-master$

Thanks and Happy New Year

milluhgram, 2017-01-27:
Do you know where to update the timezone?

chenoi, 2017-01-28:
Please check your system locale in your machine
locale -a
Make sure your enter the correct system for example ui_locale='en_US.utf8' and this is listed in your system locales machine.

tynerjim, 2017-02-07:
I'm having the exact same issue, with the unsupported locale setting.  I've checked the locale to no avail.  Does anyone have any suggestions?

dhavaljangale, 2018-02-14:
When I did $locale, I see 
LANG=
LANGUAGE=

How to add en_US.UTF-8 to locale??