HackerShackOfficial/Smart-Mirror #46: Error: 'currently'. Cannot get weather.

voodoolights, 2017-01-23:
Hello!

I ran into a problem. i get this error when running the app:
root@raspberrypi:/home/pi/Downloads/Smart-Mirror-master# sudo python smartmirror.py
Traceback (most recent call last):
  File "smartmirror.py", line 161, in get_weather
    temperature2 = "%s%s" % (str(int(weather_obj['currently']['temperature'])), degree_sign)
KeyError: 'currently'
Error: 'currently'. Cannot get weather.

Please  help me in solving this issue.
Thank You in Advance!

Mattallmighty, 2017-03-01:
Same problem too, API might have changed but I'm a noob to python

JanArneBosnes, 2017-03-13:
If its not the API, it  might be the weather_unit parameter value. 
I did a mistake by setting it to the same as the language parameter. By correcting this to:
weather_unit = 'us'
I got it running.

Pawel73, 2017-05-25:
You can set 'us' when you live in the US (Fahrenheit) or 'si' when you want to have a Celsius scale.
It is best to set 'weather_unit' = auto.
Everything should work well and the scale will adapt to the area of residence.

rdyer550, 2019-11-08:
my weather will not display! final project is due soon please help!

File "smartmirror.py", line 160, in get_weather
temperature2 = "%s%s" % (str(int(weather_obj['currently']['temperature'])), degree_sign)
KeyError: 'currently'
Error : 'currently' . Cannot get weather.

I have tried what is said in this thread and it has not fixed the problem.  Any thought would be lovely cheers!