HackerShackOfficial/Smart-Mirror #47: Cannot get weather

jamesbetterman, 2017-01-24:
this is my problem
Traceback (most recent call last):
  File "smartmirror.py", line 158, in get_weather
    weather_obj = json.loads(r.text)
  File "/usr/lib/python2.7/json/__init__.py", line 338, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 366, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 384, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
Error: No JSON object could be decoded. Cannot get weather.

J1306, 2017-02-05:
Make sure you've deleted the < > Parenthesis when inputing the api token, It then worked for me.

mauri9421, 2017-02-06:
I have the same problem!
[image]
[image]
Help please.

dhikanime, 2017-02-22:
i have the same problem, & i already remove <>. anyone help pls :<

Traceback (most recent call last):
  File "smartmirror.py", line 157, in get_weather
    weather_obj = json.loads(r.text)
  File "/usr/lib/python2.7/json/__init__.py", line 338, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 366, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 384, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
Error: No JSON object could be decoded. Cannot get weather.

mauri9421, 2017-02-22:
Probably is the api... Change it for this!
[image]
Good luck!

dhikanime, 2017-02-22:
pardon me i've already change like that before. but that make new problem :<

Traceback (most recent call last):
  File "smartmirror.py", line 160, in get_weather
    temperature2 = "%s%s" % (str(int(weather_obj['currently']['temperature'])), degree_sign)
KeyError: 'currently'
Error: 'currently'. Cannot get weather.

here's my code btw, am i did something wrong? :(

[image]
[image]
[image]
[image]
[image]
[image]

dmrowe84, 2017-03-10:
I'm seeing a lot of people with this issue, even in the closed section. 
i myself am having this issue, after browsing multiple forums i cannot for the life of me figure this out. can someone please solve this (im sure its stupidly simple) issue.
here's a screen shot of my .py file
[image]

dmrowe84, 2017-03-10:
oh and heres the error message 
[image]

dmrowe84, 2017-03-10:
@mauri9421 
Did you find a solution to the error message?

JanArneBosnes, 2017-03-13:
To dmrowe84.
You have not set the ui_locale to anything. Its on 20 and before time_format.
You can check your ui by typing this in the terminal:
locale -a

I got 'en_GB.utf8'

PabloPi1, 2017-12-23:
@JanArneBosnes I tried that and still nothing, the same message error pops up