HackerShackOfficial/Smart-Mirror #50: Attempting to make a calendar widget

TylerSporisky, 2017-02-22:
Hi,

I was tinkering with your calendar code and I can't seem to make it work.  It is to my presumption that it is unfinished.  Is this someone you could work on?

40069439, 2017-06-12:
did you manage to work it out?

tony1550, 2017-09-25:
Hi,
I was wondering the same thing and maybe getting some assistance with getting the calendar to work. I am able to see the "Calendar Events" on the mirror. I did follow the instructions in the developers google calendar quick start mentioned in the smartmirror.py script. I successfully ran the quickstart.py example, but where I am stuck is intergrading the quickstart.py in to the smartmirror.py to pull the calendar information. Can anyone help out!!

harrysprogramming, 2019-02-10:
@tony1550 could you please leave a link to the google guide as I would like to try figure this out please