HackerShackOfficial/Smart-Mirror #51: Unsupported locale setting

rockclimber396, 2017-02-26:
This is what I got when I tried to run the app.
I can't seem to find what's stopping the program from booting.

BY the way this is my first raspberry pi project. :)

pi@raspberrypi:~/Downloads/Smart-Mirror-master $ python smartmirror.py
Traceback (most recent call last):
  File "smartmirror.py", line 329, in <module>
    w = FullscreenWindow()
  File "smartmirror.py", line 306, in __init__
    self.clock = Clock(self.topFrame)
  File "smartmirror.py", line 76, in __init__
    self.tick()
  File "smartmirror.py", line 79, in tick
    with setlocale(ui_locale):
  File "/usr/lib/python2.7/contextlib.py", line 17, in __enter__
    return self.gen.next()
  File "smartmirror.py", line 38, in setlocale
    yield locale.setlocale(locale.LC_ALL, name)
  File "/usr/lib/python2.7/locale.py", line 579, in setlocale
    return _setlocale(category, locale)
locale.Error: unsupported locale setting

rockclimber396, 2017-02-26:
By the way I'm 12 years old so sorry for any stupid questions.

czoka, 2017-03-05:
I'm getting the exact same error on a Raspberry PI 3 B, would like to find a solution for it.
Also first project 😁