HackerShackOfficial/Smart-Mirror #52: Degree Sign

Pascoual92, 2017-03-03:
Hi

I can't change temperature degree sign from Fahrenheit to Celsius!
Can anyone help me?
Thanks

tomellis91, 2017-04-05:
Use "si" in Weather units. That is off the top of my head, if it doesn't work let me know and I will check mine

ashish7788, 2017-04-05:
Ok.. ill check it once nd then show you the results

ashish7788, 2017-04-05:
Hello Sir,
             I have done all changes as u say but still error in weather_obj
= json.loads(r.text)  and many more . I attach one snapshot of error report
, using this you can easy understand my point....

Thank You

tomellis91, 2017-04-05:
Can you upload your code?

ashish7788, 2017-04-05:
I have Attach all code

This is my Token : b26b5de6257ed4e1a3fb25ab840e3684

My country code IN

language : En

tomellis91, 2017-04-05:
Not the token code. A screenshot of what your editing would be useful. Can't help otherwise

tomellis91, 2017-04-05:
[](url)
[image]

tomellis91, 2017-04-05:
Weather_unit='si'

ashish7788, 2017-04-05:
Ok i understand that one , But the thing is I got weather API token so
where I put it into my code because its confusion for me . Sorry sir i
think this is irritating for you but i want to do it.

tomellis91, 2017-04-05:
weather_api_token='b26b5de6257ed4e1a3fb25ab840e3684'

tomellis91, 2017-04-05:
If you set those two fields (token and unit) and then copy everything else from the picture that i uploaded. It should work

ashish7788, 2017-04-05:
I chnage weather_api_token and weather_unit i.e. shown in screenshot

tomellis91, 2017-04-05:
Is it working? If you need any further help you have to upload the code. Otherwise it is a educated guess.

ashish7788, 2017-04-05:
Still not working only output shows clock GUI and rest of part have error.
I attach code please try it once.

# smartmirror.py
# requirements
# requests, feedparser, traceback, Pillow

from tkinter import *
import locale
import threading
import time
import requests
import json
import traceback
import feedparser
from pprint import pprint

from PIL import Image, ImageTk
from contextlib import contextmanager

LOCALE_LOCK = threading.Lock()

ui_locale = '' # e.g. 'fr_FR' fro French, '' as default
time_format = 12 # 12 or 24
date_format = "%b %d, %Y" # check python doc for strftime() for options
news_country_code = 'IN'
weather_api_token = 'b26b5de6257ed4e1a3fb25ab840e3684' # create account at https://darksky.net/dev/
weather_lang = 'en' # see https://darksky.net/dev/docs/forecast for full list of language parameters values
weather_unit = 'si' # see https://darksky.net/dev/docs/forecast for full list of unit parameters values
latitude = None # Set this if IP location lookup does not work for you (must be a string)
longitude = None # Set this if IP location lookup does not work for you (must be a string)
xlarge_text_size = 94
large_text_size = 48
medium_text_size = 28
small_text_size = 18

@contextmanager
def setlocale(name): #thread proof function to work with locale
    with LOCALE_LOCK:
        saved = locale.setlocale(locale.LC_ALL)
        try:
            yield locale.setlocale(locale.LC_ALL, name)
        finally:
            locale.setlocale(locale.LC_ALL, saved)

# maps open weather icons to
# icon reading is not impacted by the 'lang' parameter
icon_lookup = {
    'clear-day': "assets/Sun.png",  # clear sky day
    'wind': "assets/Wind.png",   #wind
    'cloudy': "assets/Cloud.png",  # cloudy day
    'partly-cloudy-day': "assets/PartlySunny.png",  # partly cloudy day
    'rain': "assets/Rain.png",  # rain day
    'snow': "assets/Snow.png",  # snow day
    'snow-thin': "assets/Snow.png",  # sleet day
    'fog': "assets/Haze.png",  # fog day
    'clear-night': "assets/Moon.png",  # clear sky night
    'partly-cloudy-night': "assets/PartlyMoon.png",  # scattered clouds night
    'thunderstorm': "assets/Storm.png",  # thunderstorm
    'tornado': "assests/Tornado.png",    # tornado
    'hail': "assests/Hail.png"  # hail
}

class Clock(Frame):
    def __init__(self, parent, *args, **kwargs):
        Frame.__init__(self, parent, bg='black')
        # initialize time label
        self.time1 = ''
        self.timeLbl = Label(self, font=('Helvetica', large_text_size), fg="white", bg="black")
        self.timeLbl.pack(side=TOP, anchor=E)
[... 234 lines cut ...]
        self.clock = Clock(self.topFrame)
        self.clock.pack(side=RIGHT, anchor=N, padx=100, pady=60)
        # weather
        self.weather = Weather(self.topFrame)
        self.weather.pack(side=LEFT, anchor=N, padx=100, pady=60)
        # news
        self.news = News(self.bottomFrame)
        self.news.pack(side=LEFT, anchor=S, padx=100, pady=60)
        # calender - removing for now
        # self.calender = Calendar(self.bottomFrame)
        # self.calender.pack(side = RIGHT, anchor=S, padx=100, pady=60)

    def toggle_fullscreen(self, event=None):
        self.state = not self.state  # Just toggling the boolean
        self.tk.attributes("-fullscreen", self.state)
        return "break"

    def end_fullscreen(self, event=None):
        self.state = False
        self.tk.attributes("-fullscreen", False)
        return "break"

if __name__ == '__main__':
    w = FullscreenWindow()
    w.tk.mainloop()

tomellis91, 2017-04-05:
ui_locale = '' # e.g. 'fr_FR' fro French, '' as default

This looks like a double quote. Delete all of that and try 
ui_locale = ' '

ashish7788, 2017-04-05:
It still not working , I will do more study and research related this one,
If it will be done the I will inform you. This is really interesting
project and wish to see your next interesting project and thank you for
giving your valuable time.

tomellis91, 2017-04-05:
Thanks, but it's not my project :) i would start again. Delete it, re-import and edit those two fields. Then it should work as expected

Pascoual92, 2017-04-07:
The point is...I dont have this code! I have this one. Can you help me with this?
 

# smartmirror.py
# requirements
# requests, feedparser, traceback, Pillow

from Tkinter import *
import time
import requests
import json
import time
import traceback
import feedparser
from PIL import Image, ImageTk

ip = ' '
country_code = 'el'
weather_api_token = 'c89cfb288ad643ada0d02fe17358c402'
time_format=24 #12 or 24

# maps open weather icons to
icon_lookup = {
    'clear-day': "assets/Sun.png",  # clear sky day
    'wind': "assets/Wind.png",   #wind
    'cloudy': "assets/Cloud.png",  # cloudy day
    'partly-cloudy-day': "assets/PartlySunny.png",  # partly cloudy day
    'rain': "assets/Rain.png",  # rain day
    'snow': "assets/Snow.png",  # snow day
    'snow-thin': "assets/Snow.png",  # sleet day
    'fog': "assets/Haze.png",  # fog day
    'clear-night': "assets/Moon.png",  # clear sky night
    'partly-cloudy-night': "assets/PartlyMoon.png",  # scattered clouds night
    'thunderstorm': "assets/Storm.png",  # thunderstorm
    'tornado': "assests/Tornado.png",    # tornado
    'hail': "assests/Hail.png"  # hail
}

class Clock(Frame):
    def __init__(self, parent, *args, **kwargs):
        Frame.__init__(self, parent, bg='black')
        # initialize time label
        self.time1 = ''
        self.timeLbl = Label(self, font=('Helvetica', 48), fg="white", bg="black")
        self.timeLbl.pack(side=TOP, anchor=E)
        # initialize day of week
        self.day_of_week1 = ''
        self.dayOWLbl = Label(self, text=self.day_of_week1, font=('Helvetica', 18), fg="white", bg="black")
        self.dayOWLbl.pack(side=TOP, anchor=E)
        # initialize date label
        self.date1 = ''
        self.dateLbl = Label(self, text=self.date1, font=('Helvetica', 18), fg="white", bg="black")
        self.dateLbl.pack(side=TOP, anchor=E)
        self.tick()

    def tick(self):
        if time_format == 12:
            time2 = time.strftime('%I:%M %p')
        else:
            time2 = time.strftime('%H:%M')
            
        day_of_week2 = time.strftime('%A')
        date2 = time.strftime("%d %b, %Y")
        # if time string has changed, update it
        if time2 != self.time1:
            self.time1 = time2
            self.timeLbl.config(text=time2)
        if day_of_week2 != self.day_of_week1:
            self.day_of_week1 = day_of_week2
            self.dayOWLbl.config(text=day_of_week2)
[... 184 lines cut ...]
        self.clock = Clock(self.topFrame)
        self.clock.pack(side=RIGHT, anchor=N, padx=100, pady=60)
        # weather
        self.weather = Weather(self.topFrame)
        self.weather.pack(side=LEFT, anchor=N, padx=100, pady=60)
        # news
        self.news = News(self.bottomFrame)
        self.news.pack(side=LEFT, anchor=S, padx=100, pady=60)
        # calender - removing for now
        # self.calender = Calendar(self.bottomFrame)
        # self.calender.pack(side = RIGHT, anchor=S, padx=100, pady=60)

    def toggle_fullscreen(self, event=None):
        self.state = not self.state  # Just toggling the boolean
        self.tk.attributes("-fullscreen", self.state)
        return "break"

    def end_fullscreen(self, event=None):
        self.state = False
        self.tk.attributes("-fullscreen", False)
        return "break"

if __name__ == '__main__':
    w = FullscreenWindow()
    w.tk.mainloop()

Pascoual92, 2017-04-07:
Basicly i want to convert fahrenheit into celsius not only the degree sign! Thanks

ashish7788, 2017-04-07:
Ok.. i will check it .. THANK-YOU

ashish7788, 2017-04-08:
I work on your code there is no error but in TK GUI no output will show , I
don't know why this is happen . I will Forward you that screenshot . check
it once. I think there is problem with fetching data from code to TKgui.

Pascoual92, 2017-04-08:
> 292618367>,

Can you send me your code? I am getting tired of this! Thanks

ashish7788, 2017-04-08:
> issues/52#issuecomment-
> 292696187>,

my python version is 3.5.3 v64

from tkinter import *
import time
import requests
import json
import time
import traceback
import feedparser
from PIL import Image, ImageTk

ip = ' '
country_code = 'el'
weather_api_token = 'c89cfb288ad643ada0d02fe17358c402'
time_format=24 #12 or 24

icon_lookup = {
'clear-day': "assets/Sun.png", # clear sky day
'wind': "assets/Wind.png", #wind
'cloudy': "assets/Cloud.png", # cloudy day
'partly-cloudy-day': "assets/PartlySunny.png", # partly cloudy day
'rain': "assets/Rain.png", # rain day
'snow': "assets/Snow.png", # snow day
'snow-thin': "assets/Snow.png", # sleet day
'fog': "assets/Haze.png", # fog day
'clear-night': "assets/Moon.png", # clear sky night
'partly-cloudy-night': "assets/PartlyMoon.png", # scattered clouds night
'thunderstorm': "assets/Storm.png", # thunderstorm
'tornado': "assests/Tornado.png", # tornado
'hail': "assests/Hail.png" # hail
}

class Clock(Frame):
        def init(self, parent, *args, **kwargs):
            Frame.__init__(self, parent, bg='black')
            # initialize time label
            self.time1 = ''
            self.timeLbl = Label(self, font=('Helvetica', 48), fg="white", bg="black")
            self.timeLbl.pack(side=TOP, anchor=E)
            # initialize day of week
            self.day_of_week1 = ''
            self.dayOWLbl = Label(self, text=self.day_of_week1, font=('Helvetica', 18), fg="white", bg="black")
            self.dayOWLbl.pack(side=TOP, anchor=E)
            # initialize date label
            self.date1 = ''
            self.dateLbl = Label(self, text=self.date1, font=('Helvetica', 18), fg="white", bg="black")
            self.dateLbl.pack(side=TOP, anchor=E)
            self.tick()

        def tick(self):
            if time_format == 12:
                time2 = time.strftime('%I:%M %p')
            else:
                time2 = time.strftime('%H:%M')
                
            day_of_week2 = time.strftime('%A')
            date2 = time.strftime("%d %b, %Y")
            # if time string has changed, update it
            if time2 != self.time1:
                self.time1 = time2
                self.timeLbl.config(text=time2)
            if day_of_week2 != self.day_of_week1:
                self.day_of_week1 = day_of_week2
                self.dayOWLbl.config(text=day_of_week2)
            if date2 != self.date1:
                self.date1 = date2
                self.dateLbl.config(text=date2)
            # calls itself every 200 milliseconds
            # to update the time display as needed
            # could use >200 ms, but display gets jerky
[... 172 lines cut ...]
        # clock
        self.clock = Clock(self.topFrame)
        self.clock.pack(side=RIGHT, anchor=N, padx=100, pady=60)
        # weather
        self.weather = Weather(self.topFrame)
        self.weather.pack(side=LEFT, anchor=N, padx=100, pady=60)
        # news
        self.news = News(self.bottomFrame)
        self.news.pack(side=LEFT, anchor=S, padx=100, pady=60)
        # calender - removing for now
        # self.calender = Calendar(self.bottomFrame)
        # self.calender.pack(side = RIGHT, anchor=S, padx=100, pady=60)

    def toggle_fullscreen(self, event=None):
        self.state = not self.state  # Just toggling the boolean
        self.tk.attributes("-fullscreen", self.state)
        return "break"

    def end_fullscreen(self, event=None):
        self.state = False
        self.tk.attributes("-fullscreen", False)
        return "break"
if __name__ == '__main__':
    w = FullscreenWindow()
    w.tk.mainloop()

Pascoual92, 2017-04-10:
I cant figure it out!! I don't know what is the problem with my code! If i try the other code there is an error with json and message can't get data from weather! That's why i asked for your code! Thanks!