HackerShackOfficial/Smart-Mirror #59: Text doesn't display fully

ajayjjain, 2017-03-24:
On my monitor, the text describing the weather and the news articles does not displays the entire String (shown in the images below). How can I fix it?
[image]
[image]

meganelizedwards, 2017-04-13:
I'm having the same problem... hmmm?

meganelizedwards, 2017-04-13:
I figured it out! You have to change the text size in the code. There's a section towards the beginning where it lists 4 different text sizes: 

xlarge_text_size = 50 
large_text_size = 25 
medium_text_size = 12 
small_text_size = 10

Those are the numbers I put in and everything fits well now. It might vary depending on the size of your monitor. 

Hope this helps!

abduls060, 2017-11-10:
Either you can change the text size or I would recommend to change the value of padding (padx and pady) under the _init_ function of the class FullscreenWindow . For all the three frames (clock, weather and news) I put padx=25 and pady=60.
It worked for me.
Hope it works for you also......!!!!

delahoya1, 2018-03-31:
my news down at the bottom doesn't show any clue why ?

closed 2021-06-18 by ajayjjain