HackerShackOfficial/Smart-Mirror #61: Please make sure you have the correct access rights and the repository exists.

lolmasterian, 2017-04-17:
Cloning into 'Smart-Mirror'...
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

bga22, 2017-04-17:
Have the same problem!
ga22@Outlook.com

chill242424, 2017-04-18:
same for me I hope the help us soon!!

therealnotorious, 2017-05-03:
also the same issue

toadfish2, 2017-05-10:
Are you cloning using SSH?  Have you setup your GitHub account with your public Keys?

AminaMokhtar, 2017-06-26:
I have the same problem

onkararj, 2017-10-25:
having same prob. can anyone tell me how to solve it?

Kashalls, 2017-11-06:
The way you would want to fix this is to use the HTTPS version of the URL instead of the SSH version.

Command: git clone <your url>

HTTPS URL: https://github.com/HackerHouseYT/Smart-Mirror.git
SSH URL: git@github.com:HackerHouseYT/Smart-Mirror.git

vijayyadava, 2019-01-04:
Select HTTPS URL from Git Hub portal as it's default URL is SSH.
Below is the same URL and connect with the same from local command line:
https://github.com/vijayyadava/AzureDevOps04JanGit.git

I got the following errors:
**Failed to push the branch to the remote repository. See the Output window for more details.
Pushing master
Error encountered while pushing branch to the remote repository: Git failed with a fatal error.
Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.**

Hope this will help you guys:)