HackerShackOfficial/Smart-Mirror #65: no $DISPLAY environment variable

skmali, 2017-05-15:
everything went smooth until I execute python smartmirror.py
I am a beginner in linux and python as well. Any help is greatly appreciated, below is the error I got:
`Traceback (most recent call last):
  File "smartmirror.py", line 330, in <module>
    w = FullscreenWindow()
  File "smartmirror.py", line 297, in __init__
    self.tk = Tk()
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1816, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: no display name and no $DISPLAY environment variable`

saikitlam3, 2017-08-02:
I ran into the same issue
and later found out can't use ssh to run the command
need to use the raspberry pi terminal
hope it helps

peacockrider, 2017-08-19:
Execute the following statement on the rpi ssh command line before executing 
python smartmirror.py

export DISPLAY=:0
This will resolve the issue.