HackerShackOfficial/Smart-Mirror #72: can anyone help me with the issue....

sunil4773, 2017-07-25:
pi@raspberrypi:~ $ python smartmirror.py
Traceback (most recent call last):
  File "smartmirror.py", line 157, in get_weather
    weather_obj = json.loads(r.text)
  File "/usr/lib/python2.7/json/__init__.py", line 338, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 366, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 384, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
Error: No JSON object could be decoded. Cannot get weather.
Traceback (most recent call last):
  File "smartmirror.py", line 235, in get_headlines
    headline = NewsHeadline(self.headlinesContainer, post.title)
  File "smartmirror.py", line 248, in __init__
    image = Image.open("assets/Newspaper.png")
  File "/usr/lib/python2.7/dist-packages/PIL/Image.py", line 2251, in open
    fp = builtins.open(fp, "rb")
IOError: [Errno 2] No such file or directory: 'assets/Newspaper.png'
Error: [Errno 2] No such file or directory: 'assets/Newspaper.png'. Cannot get news.

markferreira01, 2017-08-01:
Where did you want unzip the smart-mirror-master folder and are you running the smartmirror.py file from that location?

markferreira01, 2017-08-01:
Also have you added your API token already?