HackerShackOfficial/Smart-Mirror #73: How to make use of the news?

markferreira01, 2017-08-01:
Has anyone figured out a way to take the news headline and scroll the text? Or even better scroll the article that's tied to the headline?

markferreira01, 2017-10-12:
Ops didn't mean to close it.

PWRxPSYCHO, 2017-10-18:
To scroll the text you could create a Text widget and put all the headlines in there, and then set the size to show only 5 articles. Once you call the method again to refresh the articles you could have it scroll up to fill in the new text. However, you would want to erase the old headlines after 10 refreshes just so you are not making too many necessary variables.

markferreira01, 2017-10-19:
Hmmm...good suggestion. I'll take a look at this. Thanks for the idea!

closed 2017-10-19 by markferreira01