HackerShackOfficial/Smart-Mirror #74: Weather not fully visible

jdb0040, 2017-08-02:
[image]

See above 

Everything worked when horizontal, moved to vertical
Display_rotate=1 and loaded and looks like in the mirror. Please advise thanks!

jdb0040, 2017-08-02:
if any more information is needed please just ask. I am more than a noob (i've seen what comments people who define themselves as noobs leave) so I will do my best to answer them!

jdb0040, 2017-08-03:
when in full vertical screen but the window is opened horizontally, it works. I know I have a small screen, 19" but its all i can do for now. how to make it fit?

matthewagary, 2017-08-13:
Im having the same problem. Did you get a fix yet?

jdb0040, 2017-08-14:
Yes shorten the text!!! Had to change a lot I'll post it later. Near the top just find text sizes, change the sizes then change things from medium to small and so forth until you get it right. I even added my own event size I'll post later

markferreira01, 2017-08-14:
The issue appears to be that the text is not fitting within the frame used to generate the weather. try adding (expand = 1 , fill = "both" ) [python3 syntax] to the .pack() functions within the code. trial and error

abduls060, 2017-11-10:
Try changing values of padding(padx and pady) under _init_ function of class FullscreenWindow. It worked for me.

bervin61, 2017-12-09:
I have a solution to this where I add newlines at spaces (find the first space after 35 characters, replace it with a '\n'). I did the same thing to the news widget. Also, justify=LEFT makes the whole thing not centered.
I'd be happy to post/share code if anyone is interested

sudo21, 2018-01-04:
@bervin61 Please do!

bervin61, 2018-01-04:
I've left in some surrounding code. For the forecast:
in Weather.init
```
        self.currentlyLbl.pack(side=TOP, anchor=W)
        self.forecastLbl = Label(self, font=('Helvetica', small_text_size), fg="white", bg="black",justify=LEFT)
        self.forecastLbl.pack(side=TOP, anchor=W)
```
in Weather.get_weather
```
            forecast2 = weather_obj["hourly"]["summary"]
            start=0
            while len(forecast2[start:len(forecast2)])>40 and forecast2.count(' ',start+35):
                start=forecast2.find(' ',start+35)
                forecast2=forecast2[0:start]+'\n'+forecast2[start+1:len(forecast2)]
            icon_id = weather_obj['currently']['icon']
```

And for news:
in NewsHeadline.init
```
        self.eventName = event_name
        self.eventNameLbl = Label(self, text=self.eventName, font=('Helvetica', tiny_text_size), fg="white", bg="black",justify=LEFT,anchor=W)
        self.eventNameLbl.pack(side=LEFT, anchor=N)
```
in News.get_headlines
```
            for post in feed.entries[1:6]:
                title=post.title
                start=0
                while len(title[start:len(title)])>100 and title.count(' ',start+90):
                    start=title.find(' ',start+90)
                    title=title[0:start]+'\n'+title[start+1:len(title)]
                
                headline = NewsHeadline(self.headlinesContainer, title)
                headline.pack(side=TOP, anchor=W)
```

Adjust the hardcoded numbers to work for you. I suggest the threshold value be a bit larger than the count index (ex 100, 90)