HackerShackOfficial/Smart-Mirror #75: Problem executing "python smartmirror.py"

matthewagary, 2017-08-05:
After following all instructions and get to the final execution of "python smartmirror.py"  i get this ouput:

Traceback (most recent call last):
   File "smartmirror.py", line 14, in <module>
      from PIL import Image, ImageTk
ImportError: cannot import name ImageTk

ive tried reviewing other threads and tried numerous things. Im fairly new to Python keep in mind but have tried many things and im just at a loss as how to fix this issue to finally run "smartmirror.py" problem free!

markferreira01, 2017-10-09:
What version of Python are you using?

moaliali, 2017-10-30:
doesn't work for me

jendit, 2017-11-12:
Hi,

the smartmirror project is written in Python 2.
Make sure to call it with the number attached to the python command e. g. python2 smartmirror.py

You can check the version installed at your system with 
python --version
to show the default python version used by your system.
But as long as python 2 and 3 are installed, you can specify which to use by attaching the number to the python command. (e. g. python2 or python3).

The same goes for the pip installer commands (e.t. pip2 ...)

anubhavaich, 2017-12-06:
Says no module named tkinter.

LordBarboza, 2018-04-22:
I get this problem

Traceback (most recent call last):
  File "smartmirror.py", line 5, in <module>
    from Tkinter import *
  File "/home/pi/.linuxbrew/Cellar/python@2/2.7.14_4/lib/python2.7/lib-tk/Tkinter.py", line 39, in <module>
    import _tkinter # If this fails your Python may not be configured for Tk
ImportError: No module named _tkinter