HackerShackOfficial/Smart-Mirror #76: Adding a Sports widget

tony1550, 2017-09-25:
Hi,
Has anyone been able to add a sports feed or widget to the project? It would be nice to add like a favorite teams and score updates. If anyone has done this, please let me know. Thanks!

markferreira01, 2017-10-09:
You could probably replace the news with sports headlines, by plugging in the rss url into the get_headlines method under the News class. I haven't tried it, but I switched my mirror to pull from google.

bootyman4000, 2017-10-19:
I tried switching my newsfeed to google, but did not have any luck. I basically replaced the existing url with google's rss feed url, and it did not work. I started from scratch to see if I could get it to work properly (have yet to re-attempt it as I'm trying to get the basic page working with weather).

tony1550, 2017-10-19:
Yeah, I have tried using different RSS url's for strictly sports but none seem to work. Not sure what format it needs to be in.

keroe, 2018-10-02:
I got a sport news feed working for the german Bundesliga. The best solution is to find a sport newspaper in your language and look for their rss feed. Then you can simply exchange the url.
Otherwise you can write a web scrapper and search for a website with the scores of your favourite league and extract them from there. I did that for german handball bundesliga.