HackerShackOfficial/Smart-Mirror #77: JSON object

Marocfdshgt, 2017-10-12:
Hi guys

So I have the newest version of the repository, the program works, but I don`t get the weather. I read every single issue about the weather and also the closed ones and modified my code. Still it doesn`t work. This is the output I get:

Traceback (most recent call last):
  File "smartmirror.py", line 158, in get_weather
    weather_obj = json.loads(r.text)
  File "/usr/lib/python2.7/json/__init__.py", line 338, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 366, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 384, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
Error: No JSON object could be decoded. Cannot get weather.
root@rp:/home/pi/Smart-Mirror# 

Due to the fact that this is my topic on my half year presentation I would really appreciate any inputs and advice.

Best regards and already a big thanks

MudassirKazmi, 2017-10-12:
Have u included the API token..?

Regards,
Mudassir Kazmi

Marocfdshgt, 2017-10-12:
Yes, I added the API in the field Token, deleted < and > but left it in a string. I also  substituted  %s in the url with the token. I added my Coordinated (long and lalt) but it didnt work. Any Idea?

PWRxPSYCHO, 2017-10-12:
Where you able to get your IP location as well? There is a typo with the ip request method as well as in what is returned.

the ip request should be:

ip_json = json.loads(req.text)

NOT
ip_json = json.load(req.text)

also the return statement there seems to be an issue:

it should be 
return ip_json['ip']

NOT
return ip_json['text']

if you did not add your own coordinates for the latitude and longitude try that first to rule out the IP issue.

Marocfdshgt, 2017-10-15:
Thank you for the idea, I will anwer you asap. I am currently on holidays but in 4 days I will be back and get to try it. Sorry for the wait...

Best Regards

Marocfdshgt, 2017-10-17:
I just reinstalled everything, inserted the token and now it works... seems like a bug idk, thanks for all your help tho. !!!

closed 2017-10-17 by Marocfdshgt