HackerShackOfficial/Smart-Mirror #79: Added API Key, JSON error

bootyman4000, 2017-10-19:
I followed all the steps and added my API Key. Ran "python smartmirror.py" and received these error messages (however SmartMirror opened, displaying the date/time and news feed ONLY. No weather.):

pi@raspberrypi:~/Smart-Mirror $ python smartmirror.py
Traceback (most recent call last):
  File "smartmirror.py", line 157, in get_weather
    weather_obj = json.loads(r.text)
  File "/usr/lib/python2.7/json/__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 382, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
Error: No JSON object could be decoded. Cannot get weather.

I am generally new to this and do not no where to go from here. Google searches have turned up nothing so far. Does anyone else have this issue and/or know of a way to solve it?

tony1550, 2017-10-19:
I had that problem at one time, the way I fixed it was I had to add a latitude and longitude string located right under the api token. Replace the word "None" with your lat- long location. Go to google and look up the lat long for your city and this should fix it.

bootyman4000, 2017-10-20:
That fixed the issue! Thank you very much for the advice!

closed 2017-10-20 by bootyman4000