HackerShackOfficial/Smart-Mirror #80: Pi Zero W

bootyman4000, 2017-10-20:
Has anyone tried doing this on a pi zero w? I think my build might benefit from having the smaller form factor of the pi zero- I can fit a small speaker inside the frame so it can also play music via bluetooth (I'm going to be lazy and use the speaker's built-in bluetooth instead of programming it into the Smart-Mirror to display what song is playing. I do not have the skills or know-how to pull that off at this time).

I think the pi zero W should have no problem handling the Smart-Mirror software. I just don't want to drop a couple extra bucks on it when I already have it working on my pi3.

SpencerHalse, 2017-11-04:
Yes I can confirm that this does work. See the Image 
[image]
[image]

PabloPi1, 2017-12-22:
I as well have a Pi Zero but it is not working on mine.