HackerShackOfficial/Smart-Mirror #82: Problem with install dependencies (sudo pip install -r requirements.txt)

moaliali, 2017-10-30:
I tried to install the dependencies udo pip install -r requirements.txt), but i got the error: Could not open requirements file: (Errno 2) No such file or directory: 'requirements.txt' 
Storing debug log for failure in /root/.pip/pip.log

Im running Resbian on a virtual machine, is that a problem?

Hope to heas soon an solution.

Regards,

moaliali, 2017-10-30:
??

Kashalls, 2017-11-06:
@moaliali So what this means is, your are either A) Not in the Smart-Mirror repo folder OR B) Thaat file isn't there.

Change directory into that folder.

moaliali, 2017-11-06:
Thanks for the response @Kashalls, I already fixed it. It was B, the file wasn't there. I installed again and it's working now. Thank-you for answering!

Ahendrix9624, 2017-11-09:
@moaliali experiencing the same issue how did you install again?