HackerShackOfficial/Smart-Mirror #88: Google RSS Feed Url now out of date.

Kashalls, 2017-11-06:
Please think about making the RSS Feed Url a variable now that it is deprecated.

[image]

abduls060, 2017-11-10:
I eliminated the first post by changing the range of for loop under the function get_headlines of class News. Instead of looping for entries from  0 to 5, I iterated it from 1 to 6.

for post in feed.entries[1:6]:

It worked for me. 

Hope it helps.......!!!!!!!!

elyasnaji, 2017-11-12:
worked for me thanks man

jendit, 2017-11-12:
You are aware of that this message is basically a warning for the upcoming changes on December 1st?
http://www.kipsen.com/2017/11/google-news-to-deprecate-old-rss-feed-urls-on-december-1-2017/
I just changed my feed URL to a new one.
Just go to news.google.com, select the news you want and get an RSS URL at the bottom of the page.

abduls060, 2017-11-12:
Thanks Bro..!!!

Ahendrix9624, 2017-11-13:
@jendit I am using the link https://news.google.com/news/rss/headlines?gl=US&ned=us&hl=en for the new rss feed. Its just the headlines link but i am getting an error when trying to replace it with the old one.

abduls060, 2017-11-14:
@Ahendrix9624 Try this link https://news.google.com/news/rss/?gl=US&ned=us&hl=en
It is working for me.

Mitx01, 2017-11-19:
abduls060 link is not working for me. Can some help me please

abduls060, 2017-11-23:
@Mitx01 Which link are you talking about?

Mitx01, 2017-11-24:
I'm talking about this Link https://news.google.com/news/rss/?gl=US&ned=us&hl=en

Mitx01, 2017-11-24:
-> edit I'm also tested it with other links from different country. Maybe it's not working because I life in Switzerland! I don't know.

abduls060, 2017-11-24:
Its working for me.... Can you tell me the error what you are getting... or what really is happening when you use that link?

Mitx01, 2017-11-25:
Here is the error message:

Traceback (most recent call last):
  File "smartmirror.py", line 390, in get_headlines
    feed = feedparser.parse(headlines_url)
UnboundLocalError: local variable 'headlines_url' referenced before assignment
Error: local variable 'headlines_url' referenced before assignment. Cannot get news.

abduls060, 2017-11-25:
Check your if-else statements written before "feed = feedparser.parse(headlines_url)". I think the code is not reaching to the statement "headlines_url = "https://news.google.........................." under if-else.

So try to declare the variable headlines_url by yourself.
or just put url directly into feedparser.parse

feed = feedparser.parse("https://news.google.com/news/rss/?gl=US&ned=us&hl=en")

Mitx01, 2017-11-25:
Thanks for help @abduls060 it now working with this feed=feedparser.parse("https://news.google.com/news/rss/?gl=US&ned=us&hl=en")

closed 2017-11-27 by Kashalls