HackerShackOfficial/Smart-Mirror #98: Import request modules

analexius, 2018-01-15:
Hi guys,
everything went smoothly, but i get this when i start up:

File "smartmirror.py", line 9, in <module>
    import requests
ImportError: No module named requests

Any idea what i missed?

rgds,
Axs

nickever, 2018-01-16:
You need to install the requests package from PiPy. In terminal try:
$sudo pup install requests

-----
Nick Everett
+44 (0) 7595 630 701

analexius, 2018-01-16:
Will try tonight, thanks a mill
Axs

analexius, 2018-01-16:
(cv) pi@GrosPi:~ $ sudo pip install requests
Requirement already satisfied: requests in /usr/lib/python2.7/dist-packages
(cv) pi@GrosPi:~/Smart-Mirror $ python smartmirror.py 
Traceback (most recent call last):
  File "smartmirror.py", line 9, in <module>
    import requests
ImportError: No module named requests

nickever, 2018-01-16:
Check it’s installed for python 3 as well. 

$ pip3 freeze
(This will show a list of installed packages)

If not installed for python 3, then;
$ sudo pip3 install requests

Good luck

-----
Nick Everett
+44 (0) 7595 630 701

bervin61, 2018-01-18:
On the same lines as nickever, you can check what version of python you are running by (from the python shell)
`import sys
sys.version`
if it's installed for your version, we need to troubleshoot.