HackerShackOfficial/Smart-Mirror #100: Traceback error

matehubert, 2018-01-20:
Hi,

I got this message when is start python smartmirror.py:

pi@raspberrypi:~/Smart-Mirror $ python smartmirror.py
Traceback (most recent call last):
  File "smartmirror.py", line 313, in <module>
    w = FullscreenWindow()
  File "smartmirror.py", line 290, in __init__
    self.clock = Clock(self.topFrame)
  File "smartmirror.py", line 50, in __init__
    self.timeLbl = Label(self, font=('Helvetica', large_text_size), fg="white", bg="black")
NameError: global name 'large_text_size' is not defined
 How can i define 'large_text_size?

Thanks for any answers!

nickever, 2018-01-20:
It's already defined at the top of the smartmirror.py script:
line 29: large_text_size = 48

matehubert, 2018-01-20:
so the problem is that is defiend but on a wrong name?

matehubert, 2018-01-20:
Thanks,

i have python 3.5.3. and 2.7.13. installed! i tried to run : python2 smartmirror.py , but i  got the same error!