HackerShackOfficial/Smart-Mirror #101: Old RSS Feeds

dylanjgraham, 2018-01-31:
top headline reads "This RSS feed URL is deprecated". Google deprecated their old RSS protocols on December 1st 2017

vuongmt, 2018-03-05:
you will need to replace the headlines_url to "https://news.google.com/news/rss/?ned=us&gl=US&hl=en"

vuongmt, 2018-07-05:
here's my news code:

class News(Frame):
    def __init__(self, parent, *args, **kwargs):
        Frame.__init__(self, parent, *args, **kwargs)
        self.config(bg='black')
        self.title = 'News' # 'News' is more internationally generic
        self.newsLbl = Label(self, text=self.title, font=('Helvetica', medium_text_size), fg="white", bg="black")
        self.newsLbl.pack(side=TOP, anchor=W)
        self.headlinesContainer = Frame(self, bg="black")
        self.headlinesContainer.pack(side=TOP)
        self.get_headlines()

    def get_headlines(self):
        try:
            # remove all children
            for widget in self.headlinesContainer.winfo_children():
                widget.destroy()
            if news_country_code == None:
                headlines_url =  "https://news.google.com/news/rss/?ned=us&gl=US&hl=en"
            else:
                headlines_url = "https://news.google.com/news/rss/?ned=us&gl=US&hl=en"
                # Old news: "https://news.google.com/news?ned=%s&output=rss" % news_country_code
                
            feed = feedparser.parse(headlines_url)

            for post in feed.entries[0:5]:
                headline = NewsHeadline(self.headlinesContainer, post.title)
                headline.pack(side=TOP, anchor=W)
        except Exception as e:
            traceback.print_exc()
            print ("Error: %s. Cannot get news." % e)

        self.after(600000, self.get_headlines)