HackerShackOfficial/Smart-Mirror #103: First git clone command not working.

shawdavid, 2018-02-25:
Hello all, 
I am attempting to clone the repository using the “git clone git@github.com:HackerHouseYT/Smart-Mirror.git” command. When i run this command I receive the following message:

Permission Denied(public key)
fatal: Could not read from remote repository 

Please make sure you have the correct acces rights
And the repository exist. 

Can anyone explain why this might be happening and what I can do to successfully clone the repository.

developingjan, 2018-02-25:
Try to download the repo and install manually.

Best regards.

Michaelstone007, 2018-06-09:
git clone https://github.com/HackerShackOfficial/Smart-Mirror
or          ↑
            git
But i think the first was the good for me.
(Working with Raspbian)

birenpanchal, 2019-04-21:
i have also face this king of problem 
fatal: Could not read from remote repository
is it ok to use "git clone https://github.com/HackerShackOfficial/Smart-Mirror"
if i  want  to use first one than which step i want to follow