HackerShackOfficial/Smart-Mirror #107: Invalid Syntax

izzybella49, 2018-07-12:
My Smart Mirror gave me this message when I tried to run it:
pi@raspberrypi:~/Smart-Mirror $ python smartmirror.py
  File "smartmirror.py",  line 130
    except Exception as e:

SyntaxError:  invalid syntax

Thanks in advance

izzybella49, 2018-07-16:
def get_ip(self):
        try:
            ip_url = "https://ipstack.com"
            req = requests.get(ip_url)
            ip_json = json.loads(req.text)
            return ip_json['75.148.185.113']
            except Exception as e:

                   traceback.print_exc(e)
            return "Error: %s. Cannot get ip." % e

This is the code can someone please spot something wrong with my code.

big1surg, 2018-08-10:
def get_ip(self):
        try:
            ip_url = "http://jsonip.com/"
            req = requests.get(ip_url)
            ip_json = json.loads(req.text)
            return ip_json['ip']
        except Exception as e:
            traceback.print_exc()
            return "Error: %s. Cannot get ip." % e

dangercrow, 2019-07-06:
https://docs.python.org/2/tutorial/errors.html

Sounds like you're using python 3 syntax on python 2 code.

As the above link says, it's `catch Exception, e:` in old python