HackerShackOfficial/Smart-Mirror #108: Cannot Get Weather, latitude

elijamesn01, 2018-07-18:
Upon startup of the program, I receive this error message and the weather does not appear on the monitory.

Traceback (most recent call last) : 
    File "smartmirror.py", line 145, in get_weather
        lat = location_obj[ 'latitude' ]
KeyError: 'latitude'
Error: 'latitude'. Cannot get weather.

After tinkering with the code for several hours I am at a loss. 
Any help is appreciated!

sajuran123, 2018-07-30:
Hey

Try to use this link to find your latitude and longtitude: http://www.mapcoordinates.net/en
Type the numbers in and you will recieve the right weather data.

Best regards

Sajuran

toottRenat, 2018-08-05:
Current version of weather API is depricated, so SmartMirror needs to be updated.
Here is message that you get as dictionary location_obj:
```
0 #################################################################################################################################
1 #                                                                                                                               #
2 # IMPORTANT - PLEASE UPDATE YOUR API ENDPOINT                                                                                   #
3 #                                                                                                                               #
4 # This API endpoint is deprecated and has now been shut down. To keep using the freegeoip API, please update your integration   #
5 # to use the new ipstack API endpoint, designed as a simple drop-in replacement.                                                #
6 # You will be required to create an account at https://ipstack.com and obtain an API access key.                                #
7 #                                                                                                                               #
8 # For more information on how to upgrade please visit our Github Tutorial at: https://github.com/apilayer/freegeoip#readme      #
9 #                                                                                                                               #
a #################################################################################################################################
```

There 0-a are keys and text after them represents values. To get this message add following code before error line:
```python
for k, v in location_obj.items():
    print(k, v)
```

Ahendrix9624, 2018-09-08:
After typing in your new code I am getting the error message

Traceback (most recent call last) :
File "smartmirror.py", line 146, in get_weather
lat = location_obj[ 'latitude' ]
KeyError: 'latitude'
Error: 'latitude'. Cannot get weather.

keroe, 2018-10-02:
Hi 
In line 26&27 you will find the parameters latitude & longitude.
It looks like this:
`latitude = None`  
`longitude = None `
You need to change those parameters to your location. You can use the link sajuran provided above.
For example for New York it should look like this:
`latitude = '40' `
`longitude = '-73' `

Ahendrix9624, 2018-10-03:
@keroe @sajuran123 @elijamesn01 @toottRenat 
Even changing the lat and long in line 26 and 27 I am now getting this error
Tried researching everything please help Thanks!

pi@pi:~/Smart-Mirror $ python smartmirror.py
Traceback (most recent call last):
  File "smartmirror.py", line 159, in get_weather
    weather_obj = json.loads(r.text)
  File "/usr/lib/python2.7/json/__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 382, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
Error: No JSON object could be decoded. Cannot get weather.

toottRenat, 2018-10-04:
@Ahendrix9624 
Try using code from my pull request (it uses new API for getting weather), as least it works fine for me.
Or supply some more information with different prints like 
```
for k, v in location_obj.items():
    print(k, v)
```
or maybe printing r.text.

Ahendrix9624, 2018-10-04:
@toottRenat I think my string format is wrong for the location_req_url
I typed it just like your pull request and used my own api link this type 

Traceback (most recent call last):
  File "smartmirror.py", line 141, in get_weather
    location_req_url = 'http://api.ipstack.com/104.137.46.10?access_key=b8c717cddda21de63b6a47735aaadcda' % (self.get_ip(),ip_stack_key)
TypeError: not all arguments converted during string formatting
Error: not all arguments converted during string formatting. Cannot get weather.

toottRenat, 2018-10-05:
Yes, your string format is wrong. In your string you should replace ip and key by "%s", or completely remove string format (% and everything after it).

MitchNetz, 2019-02-01:
@toottRenat's fix works great for me!

joeleonjr, 2023-08-30:
Hi! I'm a security researcher @TruffleSecurity. I found a valid API key leaked in a comment in this Issue. I highly recommend rotating the API key immediately (not just deleting the comment). Here's a blog post I wrote on remediating these types of vulnerabilities: https://trufflesecurity.com/blog/remediate-leaked-api-keys-with-key-rotation.<br><br>If you can't find the leaked API key, you can use the following [TruffleHog](https://github.com/trufflesecurity/trufflehog) command to automatically scan for it:<br>`trufflehog github --repo https://github.com/<your_username>/<your_reponame> --issue-comments --only-verified`<br><br>If it's not clear where/what the security issue is, please let me know. I'd be happy to help. Please note that the API key might be in the past edit of a comment.<br><br><br><br>(cc @Ahendrix9624)