HackerShackOfficial/Smart-Mirror #109: Implementing Google Calender issue..

sajuran123, 2018-07-30:
Hey

Did anyone implement the "**Google Calender**" yet? 

I personally got the calendar title on the screen and the API up running from the "**Python Quickstart**". 

My question is how to you get the API from "**Python Quickstart**" implemented in the exisiting code from HackerShack?

Thank you

Sajuran

big1surg, 2018-08-08:
https://github.com/big1surg/smart_mirror/blob/master/module_calendar.py

i broke it down into a module and then just called it from the main page. its not the best way to do it but it works

        # calender, added this to the main frame
        self.calender = module_calendar.Calendar(self.bottomFrame)
        self.calender.pack(side = LEFT, anchor=S, padx=50, pady=50)

[image]

closed 2018-08-11 by sajuran123

harrysprogramming, 2019-03-12:
hi is there any way you could update your module please because the api client is discontinued