HackerShackOfficial/Smart-Mirror #113: Code is requesting for the weather too often

amdurjared, 2019-03-27:
When I run python smartmirror.py, I get this error:
ConnectionError: HTTPSConnectionPool(host='api.darksky.net', port=443): Max retries exceeded with url: /forecast/fbc3a6828ba486b19cf251cdaf8a9528/33.91752335,-84.4178225556?lang=en&units=us (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x75718d90>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',))
Error: HTTPSConnectionPool(host='api.darksky.net', port=443): Max retries exceeded with url: /forecast/fbc3a6828ba486b19cf251cdaf8a9528/33.91752335,-84.4178225556?lang=en&units=us (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x75718d90>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)). Cannot get weather.

I think the problem is that the code requests the weather too often and exceeds the limit of requests on free accounts on dark sky. Is there any way to limit the amount of times the weather is requested?

closed 2019-03-27 by amdurjared

jessicasummer, 2019-03-27:
you probably shouldve connected to the wifi

jessicasummer, 2019-03-27:
or maybe asked the experts