HackerShackOfficial/Smart-Mirror #116: SyntaxError: Missing parentheses in call to 'print'

davhan23, 2019-06-10:
Hello, I'm trying to run the SmartMirror Py for the first time and I'm getting this error. (line 201-202) I am new to this. Please help.

                if location2 == ", ":
                    self.location = "Cannot Pinpoint Location"
                    self.locationLbl.config(text="Cannot Pinpoint Location")
                else:
                    self.location = location2
                    self.locationLbl.config(text=location2)
        except Exception as e:
            traceback.print_exc()
            print "Error: %s. Cannot get weather." % e

        self.after(600000, self.get_weather)

dangercrow, 2019-07-06:
Disclaimer, have not used this project at all.

However, that sounds awfully like you're using python 3 and the project is written in python 2.

Try using `python` from the command line instead of `python3`, if you're using the latter. If not, it gets a bit more complicated - I'd suggest googling "changing default python for <your OS>"

harishMadhavan1010, 2021-05-17:
In python 2, print statement did not require parantheses but in python 3, it is more like a function and require "( )". Just use ( ) and then change imports from Tkinter to tkinter. Then the code should run just fine in python 3 as well.