HackerShackOfficial/Smart-Mirror #120: Problem during Install your dependencies

rkalra4, 2019-12-08:
[image]
[image]

I'm getting an error while installing the dependencies. I've looked all the other threads with the same issue but nothing helped me. So, can someone kindly help me with my issue.
I've also attached the screenshot showing that I do have requirements.txt file but still I'm having the issue.

Arbazkhan4712, 2019-12-08:
Use cd Smart Mirror then use pip install

rkalra4, 2019-12-08:
> Use cd Smart Mirror then use pip install

Thanks a lot for the help

closed 2019-12-08 by rkalra4